HiPERCAM/hcam-drivers #72: YSTART in GUI coords missing from FITS headers

vikdhillon, 2017-10-21:
There are no GUI-coordinate versions of HIERARCH ESO DET WIN1 YS and HIERARCH ESO DET WIN2 YS in the FITS headers. These values are 1 less than the GUI ones.

StuartLittlefair, 2017-10-21:
Good call! Any idea what we should call these? 

To make the GUI coords easier to find I followed the same ```ESO DET WIN1 XXX``` notation as the detector-output coords used by Naidu. This makes choosing a name for ```YS``` tricky since the name should make clear which to use. 

I could also switch to another naming convention...

vikdhillon, 2017-10-21:
How about something crap like HIERARCH ESO DET WIN1 YS GUI?

StuartLittlefair, 2017-10-21:
I can handle the crap-ness of that, but should I rename all the other GUI parameters as well to make them consistent, but crap?

StuartLittlefair, 2017-10-25:
fixed in https://github.com/HiPERCAM/hcam-drivers/commit/ccf01379e540daf64b44757cd87b071b5e5966cf

I kept the others as they were, and only had one crap header

closed 2017-10-25 by StuartLittlefair