HiPERCAM/hcam-drivers #76: NSKIPS not set when copying setup from fits file

vikdhillon, 2017-10-22:
We tried the new hdriver 'copying setup from fits file' option and it worked well, apart from not copying the NSKIPS we had set. This is what the fits headers show:

HIERARCH ESO DET NSKIPS      = 0 / No of skip                                   
HIERARCH ESO DET NSKIPS1     = 0 / No of skip                                   
HIERARCH ESO DET NSKIPS2     = 0 / No of skip                                   
HIERARCH ESO DET NSKIPS3     = 1 / No of skip                                   
HIERARCH ESO DET NSKIPS4     = 2 / No of skip                                   
HIERARCH ESO DET NSKIPS5     = 2 / No of skip                                   

But the GUI set them all to 1. Note also that the fits headers stores NSKIPS-1 - should we also have a GUI version of this parameter in the file headers?

StuartLittlefair, 2017-10-23:
fixed by https://github.com/HiPERCAM/hcam-drivers/commit/fdd18cbbcd274261c4d725060e88324c38c73a08

StuartLittlefair, 2017-10-23:
Note I've also closed this. I don't think it's necessary to store a separate version in the headers. The GUI knows how to translate and it shouldn't be too hard for a human to get from NSKIPS to NMULT...

closed 2017-10-23 by StuartLittlefair