HiPERCAM/hcam-drivers #77: Exposure times in the FITS headers

vikdhillon, 2017-10-22:
I might be wrong, but I can't find either the cycle time in the FITS headers, nor the user-entered exposure delay. I think it would be good to have a record of both in the headers, if possible.

vikdhillon, 2017-10-22:
Further to the above, it would be good to clarify what times we would like to store in the headers. I think we should have: the user-entered exposure delay, the actual exposure delay Naidu uses, the actual cycle time and the actual exposure time. These are for the user. Then in addition to this, we have the frame-transfer time, readout time, etc, that we need for the pipeline data reduction timestamping calculation.

At the moment, I think we do have the actual exposure time: header['EXPTIME']. I don't think we have the other three values.

StuartLittlefair, 2017-10-23:
Worth discussing this to get it right. Currently the headers contain:

- actual exposure delay used by Naidu (```DET TEXPOSE```);
- actual exposure time (``` DET SEQ1 DIT``` or ```EXPTIME```);
- time taken clearing the CCD (```DET TWIPE```);
- time take for reading the CCD (FT time plus READOUT - ```DET TREAD```);
- frame transfer time (```DET TFT```);
- other info for the pipeline, if drift mode (```DET TLINESHIFT```, ```DET TLINEDUMP```).

If I understand you correctly, then we would want to add the exposure delay requested and the cycle time. What do you think of ```DET  TEXPOSE REQUEST``` and ```DET CYCLE``` for these header items? I will need to check I'm calculating the cycle time correctly for each mode, as this will need to be done in the sequencer files in the [ESO S/W](https://github.com/HiPERCAM/hipercamcfg)

closed 2017-10-26 by StuartLittlefair