HiPERCAM/hcam-drivers #79: hcam_gain.py can't compare to historic values or write to dbase

vikdhillon, 2017-10-22:
The gain calculation script hcam_gain.py can't compare to historic values or write the results to the qc dbase.

StuartLittlefair, 2017-10-22:
Any more details on this? Perhaps an error message? It works fine for me.

vikdhillon, 2017-10-22:
Sorry! Here you go...

CCD 5, window G1: Bias Level = 1434.7, Flat level = 23637.0, RNO (ADU) = 2.7, RNO (e-) = 5.15, Gain (e-/ADU) = 1.93
CCD 5, window H1: Bias Level = 1486.9, Flat level = 20959.8, RNO (ADU) = 2.8, RNO (e-) = 5.22, Gain (e-/ADU) = 1.86

Compare to historical data (y/n): y
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pandas/io/sql.py", line 1404, in execute
    cur.execute(*args)
sqlite3.OperationalError: no such table: gain

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "hcam_gain.py", line 259, in <module>
    measure_gain()
  File "hcam_gain.py", line 241, in measure_gain
    dbf = pd.read_sql('select * from gain', conn)
  File "/usr/local/lib/python3.5/dist-packages/pandas/io/sql.py", line 400, in read_sql
    chunksize=chunksize)
  File "/usr/local/lib/python3.5/dist-packages/pandas/io/sql.py", line 1439, in read_query
    cursor = self.execute(*args)
  File "/usr/local/lib/python3.5/dist-packages/pandas/io/sql.py", line 1416, in execute
    raise_with_traceback(ex)
  File "/usr/local/lib/python3.5/dist-packages/pandas/compat/__init__.py", line 344, in raise_with_traceback
    raise exc.with_traceback(traceback)
  File "/usr/local/lib/python3.5/dist-packages/pandas/io/sql.py", line 1404, in execute
    cur.execute(*args)
pandas.io.sql.DatabaseError: Execution failed on sql 'select * from gain': no such table: gain
hipercam{observer}71:

StuartLittlefair, 2017-10-22:
Ok. I think i understand the problem. Try running through once, adding to the database but not comparing to historical data.

vikdhillon, 2017-10-22:
That worked - thanks!

closed 2017-10-22 by vikdhillon