HiPERCAM/hcam-drivers #80: Some 2-quad mode setups don't start taking data when START button is pressed

vikdhillon, 2017-10-22:
I'm too short of time to study this one in detail now, but a general observation is that the 2-quad mode is flakey - the GUI doesn't seem to start the exposure when requested and just sits there. One can make it work by changing settings (but not always). Switching to one quad always works, as does full frame. Sorry for the lack of detail here, but I wanted to get this one down in writing before I forget about it.

StuartLittlefair, 2017-10-23:
OK - worth knowing about and one to test in Edinburgh. In future, the kind of details I'd be interested in knowing about this when it happens:

- what, if anything, appears in the log windows at the bottom of hdriver
- what, if anything, appears in the hserver window

StuartLittlefair, 2017-10-24:
Just making a note to myself here, I have hunch that we might be asking for impossible parameter combinations - if you didn't look at the response log you'd miss this.

StuartLittlefair, 2018-10-17:
Havent managed to recreate this. I suspect it was caused by requesting illegal parameters. Much more logic checking has been introduced since this issue was raised.

closed 2018-10-17 by StuartLittlefair