HiPERCAM/hcam-drivers #98: Target names are all WASP-104

trmrsh, 2018-02-12:
All the target names in the FITS header seem to be recorded as WASP-104

StuartLittlefair, 2018-02-12:
Maddening @trmrsh! This is because the target name set in the telescope server is taking precedence. I had thought I had fixed this so that it did not, but obviously that fix is not working.

StuartLittlefair, 2018-02-12:
Maybe worth asking why the TO has not entered a target name, or has the wrong target name....

StuartLittlefair, 2018-02-12:
fixed in https://github.com/HiPERCAM/hcam-drivers/commit/f8fa413e569b50a40c102df0c04aff76a15f60ed

closed 2018-02-12 by StuartLittlefair

trmrsh, 2018-02-12:
Similar issue perhaps with observer (set to 'SA') and PI (set to 'gchen')

StuartLittlefair, 2018-02-12:
Yes - this is now fixed see above

trmrsh, 2018-02-12:
but I need to update hcam-drivers on the drpc do I?

StuartLittlefair, 2018-02-12:
no, i've just updated the s/w on the rack, so stopping and re-starting the GUI and hserver *should* fix this