HiPERCAM/hcam-drivers #100: Whole system hangs on GTC

StuartLittlefair, 2018-02-13:
Vik has reported the entire system can hang and not respond to commands.

This appears to be caused during an attempt to set the idle mode running. The typical command log sequence looks something like this:

```
2018-02-13 05:35:30 CMM INFO    execCommand, command = "start"
2018-02-13 05:35:31 RSP INFO    Server response = {"MESSAGEBUFFER": "OK", "RETCODE": "OK"}
2018-02-13 05:35:31 CMM INFO    Response from server was OK
2018-02-13 05:36:25 CMM INFO    execCommand, command = "abort"
2018-02-13 05:36:25 RSP INFO    Server response = {"MESSAGEBUFFER": "OK", "RETCODE": "OK"}
2018-02-13 05:36:25 CMM INFO    Response from server was OK
2018-02-13 05:36:26 CMM INFO    Adding TCS table data to run0039.fits
2018-02-13 05:36:26 CMM INFO    Response from server was OK
2018-02-13 05:36:26 CMM INFO    Run stopped
2018-02-13 05:36:26 CMM INFO    Setting chips to idle
2018-02-13 05:36:28 CMM WARNING Server response was not OK
2018-02-13 05:36:28 CMM WARNING Failed to enable idle mode
2018-02-13 05:36:28 CMM WARNING postJSON returned false
2018-02-13 05:36:28 CMM INFO    Stopping offsets (if running
2018-02-13 05:36:28 RSP INFO    Nod Server Response: {"status": "OK", "action": "stop"}
2018-02-13 05:39:06 CMM INFO    Checking M13 in simbad
2018-02-13 05:39:07 CMM INFO    M13 verified OK in simbad
2018-02-13 05:39:07 CMM INFO    Primary simbad name = M  13
2018-02-13 05:39:19 CMM WARNING Server response was not OK
2018-02-13 05:39:19 CMM WARNING Failed to post data to servers
```

Note the offending warning at 05:36:28. After this the server appears to stop responding to commands and always returns ```False``` from postJSON.

StuartLittlefair, 2018-02-14:
After enabling more debug information in the logs, I now understand the problem, but don't know how to fix it. When stopping a run, the following steps are taken:

1. Run is stopped.
2. Attempt to set chips idling - set mode to full frame 
3. Start sequencer

This fails on step 2. Setting the mode means setting the ```DET.READ.CURID``` parameter. When you do this, the acquisition task is relaunched with new parameters, and this is failing. See the error message below:

```2018-02-14 21:55:50 CMM DEBUG   Entering postJSON
2018-02-14 21:55:50 CMM DEBUG   Server URL = http://localhost:5000/setup
2018-02-14 21:55:50 CMM DEBUG   content length = 28
2018-02-14 21:55:51 RSP DEBUG   Server response: {"MESSAGEBUFFER": "failed to set DET.READ.CURID - error launching acquisiton process 1 - cannot launch process <hiperCamCCD> on ngclin7 - command i/o error", "RETCODE": "NOK"}
2018-02-14 21:55:51 CMM WARNING Server response was not OK```

So the issue is clearly one of relaunching the acquisition task when the mode is changed. I don't know why this is failing, or what the error message means. @vikdhillon if I can't work it out, we may need to contact Gao about this.

StuartLittlefair, 2018-02-14:
@eh2006re Do you have any ideas about the error message above? It comes when we change readout mode using 

```
ngcbCmd setup DET.READ.CURID 1
```

for example...

vikdhillon, 2018-02-14:
Problem seems far worse tonight than last night - I'm having to restart start_hicam after every successful exposure. Anything fudgey we can do to get it going for tonight?

StuartLittlefair, 2018-02-15:
May have a fix for this, inspection of log output from ```hiperCamCCD``` showed that the acquisition task was being launched with demultiplexing on, but the number of binned y pixels was not a multiple of 4 causing the acquisition task to fail. 

If this is correct, this is fixed in https://github.com/HiPERCAM/hipercamcfg/commit/32e36ad622364317cba21c570ea741eb25463b69

StuartLittlefair, 2018-02-15:
for future reference, ```hipercamCCD``` log output can be viewed with ```journalctl | grep hiperCamCCD```

closed 2018-02-15 by StuartLittlefair