HiPERCAM/hcam-drivers #101: Fileserver can't serve files until 3 frames appear

StuartLittlefair, 2018-02-13:
```
hipercam{observer}126: rtplot
rrun - run name [run0031]: run0041
first - first frame to plot [0]: \
Traceback (most recent call last):
  File "/home/observer/.local/bin/rtplot", line 11, in <module>
    load_entry_point('hipercam==0.7.dev4+g61ea9d5', 'console_scripts',
'rtplot')()
  File "/home/observer/.local/lib/python3.5/site-packages/hipercam/scripts/rtplot.py",
line 452, in rtplot
    for mccd in spool:
  File "/home/observer/.local/lib/python3.5/site-packages/hipercam/spooler.py",
line 265, in __next__
    return self._iter.__next__()
  File "/home/observer/.local/lib/python3.5/site-packages/hipercam/hcam.py",
line 782, in __next__
    return self.__call__()
  File "/home/observer/.local/lib/python3.5/site-packages/hipercam/hcam.py",
line 853, in __call__
    raw_bytes = self._ws.recv()
  File "/usr/local/lib/python3.5/dist-packages/websocket/_core.py",
line 296, in recv
    opcode, data = self.recv_data()
  File "/usr/local/lib/python3.5/dist-packages/websocket/_core.py",
line 313, in recv_data
    opcode, frame = self.recv_data_frame(control_frame)
  File "/usr/local/lib/python3.5/dist-packages/websocket/_core.py",
line 326, in recv_data_frame
    frame = self.recv_frame()
  File "/usr/local/lib/python3.5/dist-packages/websocket/_core.py",
line 360, in recv_frame
    return self.frame_buffer.recv_frame()
  File "/usr/local/lib/python3.5/dist-packages/websocket/_abnf.py",
line 340, in recv_frame
    self.recv_header()
  File "/usr/local/lib/python3.5/dist-packages/websocket/_abnf.py",
line 288, in recv_header
    header = self.recv_strict(2)
  File "/usr/local/lib/python3.5/dist-packages/websocket/_abnf.py",
line 375, in recv_strict
    bytes_ = self.recv(min(16384, shortage))
  File "/usr/local/lib/python3.5/dist-packages/websocket/_core.py",
line 430, in _recv
    return recv(self.sock, bufsize)
  File "/usr/local/lib/python3.5/dist-packages/websocket/_socket.py",
line 94, in recv
    "Connection is already closed.")
websocket._exceptions.WebSocketConnectionClosedException: Connection
is already closed.
hipercam{observer}127:
```

And the fileserver window reports the error:

```
Connection opened to access run0041
Socket closed
Connection opened to access run0041
ERROR:tornado.application:Uncaught exception in /run0041
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/websocket.py",
line 494, in _run_callback
    result = callback(*args, **kwargs)
  File "/usr/bin/fileserver", line 68, in on_message
    self.get_last_frame()
  File "/usr/bin/fileserver", line 95, in get_last_frame
    self.ffp.seek_frame(self.ffp.num_frames - 1)
  File "/usr/lib/python2.7/site-packages/hcam_drivers/utils/web.py",
line 97, in seek_frame
    self._fileobj.seek(self.header_bytesize + self.framesize*(frame_number-1))
IOError: [Errno 22] Invalid argument
Socket closed
Connection opened to access run0041
Socket closed
Connection opened to access run0041
Socket closed
```

StuartLittlefair, 2018-02-13:
@trmrsh @vikdhillon 

I believe this is an issue involving the calculation of the number of frames in the fileserver. Have lodged this issue to keep track.

StuartLittlefair, 2018-02-13:
@trmrsh I think I'm right in suspecting issues with how we calculate the frame size and the number of frames as the file is still growing. However, things are not simple to fix, because I'm puzzled by the way the file size grows and the relation to the frame size.

**example: FF data**
total framesize = 36 +  2 * (1024 * 2048 * 5) = 20971556
image size without timestamp (bytes)  = 2 * (1024 * 2048 * 5) = 20971520

Then if I poll the actual file size with ```os.stat``` it follows this pattern:

```
20971520
41943040
62914560
83886080
...
```

which are integer multiples of the frame size *without the timing bytes*. I am very confused by this. Any ideas?

trmrsh, 2018-04-07:
Could you poll the ESO NGC to see how many frames it thinks have been taken? I know it's nicer to work from the file, but it seems difficult to do so (in fact I suspect impossible based on filesize alone).

tom

StuartLittlefair, 2018-04-07:
Yes, that’s possible. However I’m quite keen to keep the fileserver separate from the eso server if *at all* possible. There are plenty of occasions where you might want to run the fileserver without running the eso server.

trmrsh, 2018-04-07:
yes, sure. At the very least you would need a fallback is the ESO server wasn't on. Is it possible to update any of the header items in the FITS file on the fly? Alternatively you could work out the number of frames exposed according to how long the exposure has been going. Again not nice. Are the unwritten bytes all set to zero? One could use that perhaps? Or attempt to read the timing bytes at the end of the frames. All pretty horrible.

StuartLittlefair, 2018-04-13:
So the frame number we are requesting is not the issue, but it is related to the filesize on the disc, which I do not understand (see [this comment](https://github.com/HiPERCAM/hcam-drivers/issues/101#issuecomment-365333271)).

At the moment we cannot grab frame 1 because:

- The header size is X bytes. The framesize without timing bytes is Y. The framesize including timing bytes is Z.
- the algorithm to grab frame 1 is to move the file read pointer to X, and try to read Z bytes.
- this fails when only frame 1 is written, because the file size is only X+Y bytes
- the batshit crazy thing is that once frame 2 *is* written, I can read Z bytes and the data seems to be OK, i.e there are timing bytes available. 

In summary @vikdhillon, @trmrsh I really don't understand how the data frames are being written and what is in them over time, which is making this hard to fix. I'm going to save some snapshots of the data as it is being written, so I can play offline.

trmrsh, 2018-04-13:
Sounds really bonkers and annoying. Does this mean even if you have a way of knowing that frame 1 is available, you can't read it? Damn. Hope we won't need to return to bespoke data format, although it does have its advantages. If we had to we could have the fits file just for the headers and write the data/timing to an ULTRACAM-like separate data file (yuck). I wonder if there is some way to initialise the data array to the right size for the first frame + timing bytes? (may not help though). I was going to suggest file modification time as a check of whether a frame is available, but by the sound of it, this isn't the issue.

StuartLittlefair, 2018-04-13:
@trmrsh that’s right. Even if we know frame 1 is available we can’t read it. 

In fact, I am beginning to suspect a bug in the acquisition task such that the final 36 bytes of each frame (last 18 pixels) are not getting written. I am investigating that possibility.

StuartLittlefair, 2018-04-13:
I think we've cracked this. At least for FF, slow data this is what's happening:

The pixel data is ```dsize``` bytes, and there are ```n``` timing bytes. A total frame is ```fsize = dsize + n``` bytes. The data gets written to a memory buffer and then flushed to disk. *However*, at the end of each frame, only ```dsize``` bytes are flushed to the disk. 

The consequences are:
- that frame 1 is not fully written to disk until frame 2 is ready and the buffer is flushed a second time
- the buffer is gradually filling up, which might eventually lead to data loss?

StuartLittlefair, 2018-04-13:
Since this is an acquisition task bug, I am closing this and redirecting to https://github.com/HiPERCAM/hiperCamCCD/issues/5

closed 2018-04-13 by StuartLittlefair