HiPERCAM/hcam-drivers #110: Have to click START twice to get drift mode run started

vikdhillon, 2018-05-20:
I noticed that when I switch to a drift mode setup, I have to hit START twice to get the run to actually start. On first clicking START, the exposure doesn't start and the START button stays illuminated green. I attach the output of the hserver window below. The setup that does this is the MAXI J1820 setup on May 19 (run0032).

BINY1 1  DET.NSKIPS2 0  DET.NSKIPS3 0  DET.NSKIPS4 0  DET.INCOVSCY F  DET.NSKIPS1 0  DET.NSKIPS5 0  DET.SEQ.TRIGGER F  DET.NCLRS 5  DET.FASTCLK 0  DET.BINX1 1  DET.RESTRICT F  DET.SEQ1.DIT 10  DET.DUMMY 0  DET.GPS F  DET.SEQ.CLKFILE hipercam_se.bclk  DET.TDELAY.GUI 10000 ']
['ngcbCmd', 'seq start']
['ngcbCmd', 'seq stop']
['ngcbCmd', 'setup DET.READ.CURID 4']
['ngcbCmd', 'acqproc hiperCamCCD -chip 5 -t 2 -nq 5 -deml 0 -gps 1 -aveg 1 -aps 8 -drf 1']
['ngcbCmd', 'setup M2UZ 0  DECDEG 74.484513  M2UY 0  SLATEL LP10.0  OBJECT Bias  ELEVAT 28.8  PARANG -77.806  M2RY 0  M2RX 0  WINDSPEE 3.26267 ']
['ngcbCmd', 'setup INSTRPA 0.0  OBSERVER vsd,spl  POX 0  POY 0  WINDDIRE 343  FRAME UNDEF  LST +05:43:38.51  HUMIN 29.5  INSTRUME HIPERCAM  GTCPRGID GTC04-18ADDT ']
['ngcbCmd', 'setup HUMOUT 29.6  M2UX 0  DEWPOINT -6.0615  OBSERVAT ORM  HEIGHT 2348  PRESSURE 776.297  RA 0:00:00  AZIMUTH 342.4  DOME_AZ 54.9995  PI Shahbaz ']
['ngcbCmd', 'setup AIRMASS 2.08  TELESCOP GTC  T_PRIM 8.3  GTCOBID 1  RADEG 0.00000  HUMIDITY 29.65  ROTATOR -2.22222e-05  TELFOCUS 0.0  DEC +74:29:04  LONGITUD 17.877639 ']
['ngcbCmd', 'setup ORIGIN GRANTECAN  TAMBIENT 10.0682  OBSMODE Bias  IMAGETYP bias  T_TUBE 9.71929  FILTERS us,gs,rs,is,zs  LATITUDE 28.762000  PROGRM GTC04-18ADDT  MOONDIST 82.97 ']
['ngcbCmd', 'setup CCD5TEMP -90.0  CCD3VAC 0.00077  CCD3TEMP -87.8  CCD1VAC 0.00082  CCD2VAC 0.00113  CCD5VAC 0.00093  CCD4TEMP -90.0  CCD1TEMP -89.9  CCD2TEMP -88.6  CCD4VAC 0.00098 ']
['ngcbCmd', 'setup CCD5FLOW 1.35  CCD4FLOW 1.22  CCD3FLOW 1.29  CCD1FLOW 1.29  CCD2FLOW 1.16  FPSLIDE -99.999804988 ']
['ngcbCmd', 'setup DET.DRWIN.XSH 112  DET.DUMMY 1  DET.SPEED 1  DET.DRWIN.XSR 1825  DET.BINY1 8  DET.NSKIPS2 0  DET.NSKIPS3 0  DET.NSKIPS4 0  DET.NSKIPS5 0  DET.DRWIN.YS.GUI 1  DET.FASTCLK 1  DET.BINX1 8  DET.RESTRICT F  DET.DRWIN.NY 72  DET.DRWIN.NX 96  DET.TDELAY.GUI 0.01  DET.EXPLED F  DET.INCPRSCX F  DET.CLRCCD F  DET.DRWIN.YS 0  DET.SEQ.TRIGGER F  DET.NCLRS 5  DET.DRWIN.XSG 128  DET.DRWIN.XSF 128  DET.DRWIN.XSE 112  DET.NSKIPS1 0  DET.SEQ1.DIT 0.001775  DET.INCOVSCY F  DET.GPS T  DET.SEQ.CLKFILE hipercam_fastclk.bclk  DET.DRWIN.XSL 113  DET.FRAM2.BREAK 10001 ']
['ngcbCmd', 'seq stop']
['ngcbCmd', 'setup DET.READ.CURID 4']
['ngcbCmd', 'acqproc hiperCamCCD -chip 5 -t 2 -nq 5 -deml 0 -gps 1 -aveg 1 -aps 8 -drf 1']
['ngcbCmd', 'setup M2UZ 0  DECDEG 74.484513  M2UY 0  SLATEL LP10.0  OBJECT Bias  ELEVAT 28.7  PARANG -77.9408  M2RY 0  M2RX 0  WINDSPEE 3.33907 ']
['ngcbCmd', 'setup INSTRPA 0.0  OBSERVER vsd,spl  POX 0  POY 0  WINDDIRE 322  FRAME UNDEF  LST +05:44:12.645  HUMIN 29.4  INSTRUME HIPERCAM  GTCPRGID GTC04-18ADDT ']
['ngcbCmd', 'setup HUMOUT 30.0  M2UX 0  DEWPOINT -5.7851  OBSERVAT ORM  HEIGHT 2348  PRESSURE 776.376  RA 0:00:00  AZIMUTH 342.4  DOME_AZ 54.9995  PI Shahbaz ']
['ngcbCmd', 'setup AIRMASS 2.08  TELESCOP GTC  T_PRIM 8.3  GTCOBID 1  RADEG 0.00000  HUMIDITY 29.96  ROTATOR -2.22222e-05  TELFOCUS 0.0  DEC +74:29:04  LONGITUD 17.877639 ']
['ngcbCmd', 'setup ORIGIN GRANTECAN  TAMBIENT 10.1783  OBSMODE Bias  IMAGETYP bias  T_TUBE 9.72357  FILTERS us,gs,rs,is,zs  LATITUDE 28.762000  PROGRM GTC04-18ADDT  MOONDIST 82.97 ']
['ngcbCmd', 'setup CCD5TEMP -90.0  CCD3VAC 0.00077  CCD3TEMP -87.8  CCD1VAC 0.00082  CCD2VAC 0.00112  CCD5VAC 0.00092  CCD4TEMP -90.0  CCD1TEMP -90.0  CCD2TEMP -88.6  CCD4VAC 0.00097 ']
['ngcbCmd', 'setup CCD5FLOW 1.35  CCD4FLOW 1.22  CCD3FLOW 1.29  CCD1FLOW 1.35  CCD2FLOW 1.1  FPSLIDE -99.999804988 ']
['ngcbCmd', 'setup DET.DRWIN.XSH 112  DET.DUMMY 1  DET.SPEED 1  DET.DRWIN.XSR 1825  DET.BINY1 8  DET.NSKIPS2 0  DET.NSKIPS3 0  DET.NSKIPS4 0  DET.NSKIPS5 0  DET.DRWIN.YS.GUI 1  DET.FASTCLK 1  DET.BINX1 8  DET.RESTRICT F  DET.DRWIN.NY 72  DET.DRWIN.NX 96  DET.TDELAY.GUI 0.01  DET.EXPLED F  DET.INCPRSCX F  DET.CLRCCD F  DET.DRWIN.YS 0  DET.SEQ.TRIGGER F  DET.NCLRS 5  DET.DRWIN.XSG 128  DET.DRWIN.XSF 128  DET.DRWIN.XSE 112  DET.NSKIPS1 0  DET.SEQ1.DIT 0.001775  DET.INCOVSCY F  DET.GPS T  DET.SEQ.CLKFILE hipercam_fastclk.bclk  DET.DRWIN.XSL 113  DET.FRAM2.BREAK 10001 ']
['ngcbCmd', 'start']
['ngcbCmd', 'seq stop']
['ngcbCmd', 'setup DET.READ.CURID 1']
['ngcbCmd', 'acqproc hiperCamCCD -chip 5 -t 2 -nq 5 -deml 0 -gps 1 -aveg 4 -aps 8 -drf 0']
['ngcbCmd', 'setup TELESCOP WHT  OBSMODE Imaging  RA 00:00:00.00 ']
['ngcbCmd', 'setup ORIGIN ING  AIRMASS -99  INSTRPA -99  ELEVAT -99  AZIMUTH -99  TELFOCUS -99  DEC +00:00:00.0  CCD1TEMP -99  CCD2TEMP -99  MOONDIST -99 ']
['ngcbCmd', 'setup CCD5TEMP -99  CCD3VAC -99  CCD3TEMP -99  CCD1VAC -99  CCD2FLOW -99  CCD2VAC -99  CCD1FLOW -99  CCD5VAC -99  CCD4TEMP -99  CCD4VAC -99 ']
['ngcbCmd', 'setup CCD3FLOW -99  CCD5FLOW -99  FPSLIDE -99  CCD4FLOW -99 ']
['ngcbCmd', 'setup DET.EXPLED F  DET.INCPRSCX F  DET.CLRCCD T  DET.SPEED 0  DET.BINY1 1  DET.NSKIPS2 0  DET.NSKIPS3 0  DET.NSKIPS4 0  DET.INCOVSCY F  DET.NSKIPS1 0  DET.NSKIPS5 0  DET.SEQ.TRIGGER F  DET.NCLRS 5  DET.FASTCLK 0  DET.BINX1 1  DET.RESTRICT F  DET.SEQ1.DIT 10  DET.DUMMY 0  DET.GPS F  DET.SEQ.CLKFILE hipercam_se.bclk  DET.TDELAY.GUI 10000 ']
['ngcbCmd', 'seq start']

StuartLittlefair, 2018-05-30:
Hi @vikdhillon.

The underlying issue here is that sometimes the ESO s/w takes a long time to switch to drift mode from full frame mode. I don't know why this is, but it may be something to do with the acquisition task. Really, that underlying issue needs fixing. 

However, the direct cause of the problem you see is that this delay in switches modes causes the GUI to timeout when posting the run. I can increase the timeout.

StuartLittlefair, 2018-06-01:
duplicate of #107. Closing

closed 2018-06-01 by StuartLittlefair