HiPERCAM/hcam-drivers #111: Incorrect naming conventions in hdriver

vikdhillon, 2018-05-20:
A minor thing, but there are a few naming inconsistencies in hdriver labels that it would be nice to sort out:

1. nodding - strictly speaking this refers to moving a star up and down a slit, so "dithering" would be a more correct term.

2. overscan - this is the bias strip in the row direction, and pre-scan is the bias strip in the column direction. The latter can be found on both windows and full frame, whereas the former only in full frame mode. So the label on the gui might be better called pre-scan? Also, when the user is warned about using binning factors that will merge a bit of data with bias, the pop-up window always says overscan, but often means pre-scan.

StuartLittlefair, 2018-06-01:
@vikdhillon I've cleared up the nodding/dither issue. 

W.R.T overscan/prescan, is this the kind of distinction the instrument user actually cares about? I was just using overscan as a generic term. 

As an aside, I don't think you are correct about overscan only being available in full frame mode. Certainly enabling overscan in 1 window mode causes Naidu to clock another set of lines...

vikdhillon, 2018-10-04:
Thanks. Regarding overscan in windowed mode (where I really mean overscan, i.e. overscan in y in hdriver gui terminology) - this isn't available in hdriver as an option (it is greyed out when expert level 1 is selected). This is correct, of course, as Tom gave up trying to implement it in the pipeline. Weirdly, I found an email from Naidu in Dec 17 saying that it isn't available in his code, although obviously if you can see the lines in the code for yourself, and Tom tried to implement it in the pipeline, it must be!

Regarding the terminology - to make it crystal clear to the astronomer and avoid confusing terminology for CCD experts, how about renaming overscan on the gui to "bias strip"?

One other terminology issue (perhaps): when you shut down the gui, it asks if you want to turn the clocks off? Is that correct? Isn't it actually the bias voltages? (The ones that enable the reverse bias in the p-n junctions and widens the depletion region.) The clocks are only enabled when one starts moving charge around.

closed 2023-07-27 by StuartLittlefair