HiPERCAM/hcam-drivers #114: rtplot server doesn't work with full frame

vikdhillon, 2018-05-21:
When the rtplot server is running, no data is returned when in full frame mode. Windowed mode works fine. This makes rtplot flag up an irritating error every time you plot a full frame:
** bad return from hdriver = No valid data available

StuartLittlefair, 2018-06-01:
Is this not correct behaviour? I'd argue that the rtplot server should return some sort of error string in the case of full frame mode, which is what it currently does. 

I can fix here, but my preference is for @trmrsh to fix this in rtplot. Comments @vikdhillon?

trmrsh, 2018-06-01:
Well, there clearly is valid data, namely the full frame windows. I would expect a return of "no valid data" to indicate some other problem, not that we are in a perfectly valid state, otherwise how can I tell the difference between full frame and a genuine problem? I can easily suppress the warning in rtplot, but then I might be giving the observer a false sense of security.

StuartLittlefair, 2018-06-01:
Is it ok then if I return either 'no valid data', 'fullframe' or the window parameters? I'm trying to keep my code pretty general and returning the full ccd as if it were a window would require a bit of a code restructure...

trmrsh, 2018-06-01:
can't you just return full frame parameters as if they were 4 windows? ideally rtplot should not need to assume values for the window sizes, i.e. it should not need to know that it is talking to the hipercam gui as opposed to the ultracam one.

StuartLittlefair, 2018-06-01:
I can, but I'd have to refactor the code a bit and add about 10 lines. It seems easier and more logical to me to add a two line if statement to rtplot which doesn't draw *anything* on the display if ithe server returns a string that says 'full frame'

trmrsh, 2018-06-01:
I can certainly do that; in fact it is exactly what I do at the moment with the addition of the error message to the terminal so it will be a matter of trapping that instead. Please confirm whether the return will be "full frame" or as you had it earlier "fullframe". Bear in mind that the error condition will also be to plot no setup windows.

StuartLittlefair, 2018-06-01:
Thanks tom - makes my life much easier. It will be 'fullframe', all lower case, no space.

This change added in https://github.com/HiPERCAM/hcam_widgets/commit/59f1d39930fbaf714f5712689c45a9068a6c7444

closed 2018-06-01 by StuartLittlefair