HiPERCAM/hcam-drivers #117: GTCOBID box required on GUI

vikdhillon, 2018-05-21:
Sometimes we use a saved json file in hdriver which has an OB number that is different to the one that is sent to the telescope. For example, for Boris' WD1145 program, he only entered 3 OBs in phase II, but we had to observe the target a fourth time, so we used the OBID=3 json files. This wrote GTCOBID=0003 to the fits headers, confusing the GTC archiving system. We could have hand edited the json files, but it would be much safer and easier to have an OBID box on the GUI, perhaps underneath the proposal ID box.

StuartLittlefair, 2018-05-30:
hi @vikdhillon 

I think you've reached the same conclusion with GTC staff in email discussions, but it's probably better not to include this information as a box in the GUI. I'm concerned about the user getting things wrong and putting the wrong OBID in. 

Ideally there should be a 1:1 relationship between the OBID and a JSON file so we can track back and understand errors etc. If you agree, I'll close this issue.

vikdhillon, 2018-05-30:
Hi Stu,
Yes, fine with me.
Cheers,
Vik.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

closed 2018-06-01 by StuartLittlefair