HiPERCAM/hcam-drivers #121: hdriver log file append/overwrite

vikdhillon, 2018-10-04:
When starting hdriver, it asks if you want to append to or overwrite the existing log file. Clicking append does nothing, so one has to click overwrite, which I seem to remember Stu saying doesn't actually overwrite. So couldn't we just remove this prompt window entirely? (I write this having shutdown/restarted hdriver about 100 times this afternoon, so being continually prompted for this eventually motivated me to write this issue).

StuartLittlefair, 2018-10-04:
We have no control over this sadly. It’s built into the file open dialog used by the GUI library. 

If it really bugs you the fix is for me to write my own file open dialog.

vikdhillon, 2018-10-04:
Don't worry about it!

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

closed 2018-10-08 by StuartLittlefair