HiPERCAM/hcam-drivers #129: slide does not move to position in command-line mode

vikdhillon, 2018-10-30:
When running the slide from the command line on the rack PC. The slide home command seems to work fine (reports a position of 0 microsteps and 0 mm and 2760.2 pixels). When you then type slide park or slide pos=1100px, for example, the slide only moves part of the way (e.g. a slide position returns 2371.8 pixels) without returning any error.

StuartLittlefair, 2018-10-30:
Are you sure it doesn't get there in the end? The command will return after the **estimated** time for the command to complete. 

try ```slide position``` again after a while

vikdhillon, 2018-10-30:
No it definitely doesn't ever get there (with the old version of the slide
software). I have to do repeated slide park commands to get there. The same
goes for slide home commands. If I'm parked, I have to do repeated slide
home commands to actually get home.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

vikdhillon, 2018-10-30:
Tested the slide_updates version of the slide program, on the command line.
This works better - the slide moves from home to park, or from park to
home, in one go. So that has fixed the problem of having to enter the
command multiple times.
However, the calculated time to make the move is much too short, so it
returns control to the command line before it has actually moved all the
way. Repeated slide position commands show that the slide is still moving.
Also, the slide position command no longer reports the number of microsteps
or the number of mm, only the number of pixels. (This could be because the
first part of the output string written to the screen is being overwritten
by some kind of debug message).

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

vikdhillon, 2018-10-30:
Tested the slide from hdriver now. Seems to work fine. However, we
repeatedly get these error messages popping up in the command log box and
the xterm in which hdriver was started:

15:10:15 CMM WARNING Error in thread Slide: ValueError: need more than 2
values to unpack
15:10:36 CMM WARNING Error in thread Slide: ValueError: need more than 2
values to unpack
15:10:56 CMM WARNING Error in thread Slide: ValueError: need more than 2
values to unpack
15:11:12 CMM WARNING Could not update pressure for CCD 2: could not parse
response Selected hunt group busy
15:11:15 CMM WARNING Error in thread Slide: ValueError: need more than 2
values to unpack

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2018-10-30:
That makes sense - the error is one of the widgets in the GUI checking the slide using the old method. There's a matching ```slide_updates``` branch on hcam_widgets that needs to be installed.

StuartLittlefair, 2018-10-30:
How far out is the timeout BTW?

vikdhillon, 2018-10-30:
About a factor of 4??

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2018-10-30:
So the reason for this is that I've forgotten how the slide works. A command like ```slide home``` doesn't block the terminal until the slide move is over. Instead it returns as soon as the command is successfully sent to the slide - the delay you see on the command line is the time it takes to talk to the slide (the terminal server is slow).

In other words, I think this is normal behaviour, but I could make the slide command on the terminal work a different way, more like the old slide command (i.e it will block whilst moving).

closed 2018-11-18 by StuartLittlefair