HiPERCAM/hcam-drivers #132: Problem dithering

martinjohndyer, 2018-10-31:
When taking flats hdriver the dithering doesn't work. It seems to work fine from the command line `gtc_tweak_pointing` but not `hdriver`. We get a warning that the exposure will continue indefinitely.

Has anything been changed there @StuartLittlefair?

martinjohndyer, 2018-10-31:
Also somehow a `run0000.fits` file was created, haven't seen that before. Not sure if it's connected or not. We moved it to `/data/junk/0000.fits`.

vikdhillon, 2018-10-31:
Here is the error reported in the gtc_offset window:

starting
forcing an offset
{'exc_info': (<class 'tornado.web.HTTPError'>, HTTPError(), <traceback object at 0x457f440>)}
ERROR:tornado.application:Uncaught exception in write_error
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 1036, in send_error
    self.write_error(status_code, **kwargs)
  File "/usr/bin/gtc_offsetter", line 185, in write_error
    self.finish(json_encode(resp_dict))
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 991, in finish
    self.flush(include_footers=True)
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 947, in flush
    start_line, self._headers, chunk, callback=callback)
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/http1connection.py", line 386, in write_headers
    raise ValueError('Newline in header: ' + repr(line))
ValueError: Newline in header: "HTTP/1.1 500 Failed to force offset.\n'InotifyObserver' object has no attribute 'do_next_offset'"
ERROR:tornado.application:Exception in exception handler
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 1516, in _execute
    self._handle_request_exception(e)
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 1568, in _handle_request_exception
    self.send_error(e.status_code, exc_info=sys.exc_info())
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 1040, in send_error
    self.finish()
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/web.py", line 992, in finish
    self.request.finish()
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/httputil.py", line 419, in finish
    self.connection.finish()
  File "/usr/lib/python2.7/site-packages/tornado-4.5.1-py2.7-linux-x86_64.egg/tornado/http1connection.py", line 448, in finish
    self._expected_content_remaining)
HTTPOutputError: Tried to write 444 bytes less than Content-Length
stopping
stopping

vikdhillon, 2018-10-31:
Quite urgent this one, as we can't really do the ultra-faint, ultra-diffuse galaxy observations without dithering working.

StuartLittlefair, 2018-10-31:
On my way home. Will try and fix urgently.

StuartLittlefair, 2018-10-31:
Should be fixed by 
https://github.com/HiPERCAM/hcam-drivers/commit/b19f66ccabb2dcac71fba8cb5ab659235d478caa

Please install latest master branch of hcam-drivers and hcam_widgets. You’ll need to restart the ```gtc_offsetter```. Let me know how you go - hopefully this was fixed in time

closed 2018-10-31 by StuartLittlefair

vikdhillon, 2018-11-01:
We installed and tested this last night on the dawn flats and it worked
fine. Thanks!