HiPERCAM/hcam-drivers #133: Warnings on starting hdriver

vikdhillon, 2018-11-01:
We get the following warnings when starting hdriver - thought you should see them in case there is anything worrying. Presumably we can ignore the astropy error? What about the dithering being enabled in non-clear mode - this isn't true, as hdriver opened with last night's flat field setup, which has dithering enabled with clear mode:

ngclin7 insuser:~ 1026 > 15:27:38 CMM INFO    Tel/ins = GTC
15:27:38 CMM INFO    Longitude = -17.87764 E
15:27:38 CMM INFO    Latitude = 28.762 N
15:27:38 CMM INFO    Elevation = 2300.0 m
/usr/lib64/python2.7/site-packages/astropy/time/core.py:749: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  elif longitude == 'greenwich':
15:27:41 CMM WARNING ERR: dithering enabled with clear mode off
15:27:41 CMM INFO    Loaded saved instrument and run settings
rtplot server started
15:27:58 CMM WARNING Could not update table of TCS info
15:28:06 CMM WARNING Could not update pressure for CCD 2: could not parse response Selected hunt group busy

StuartLittlefair, 2018-11-01:
Was clear mode actually enabled when hdriver started?

There's nothing serious here, but I might be saving/loading json files incorrectly

vikdhillon, 2018-11-01:
Yes, it seemed to be (box was white)

vikdhillon, 2018-11-01:
> Hmm. I'll look into it then, but I don't see how this could be a real
> issue. Do you get the same warning every time you load the flats setup?

Yes, we do!

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2018-11-01:
Ok that’s interesting. It might be to do with the order stuff appears in the JSON file. 

Can you mail me the JSON and I’ll look at this tomorrow. I can’t see this causing problems but make sure you actually spiral during flats this evening!

vikdhillon, 2018-11-01:
Attached.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---