HiPERCAM/hcam-drivers #134: OB box on hdriver

vikdhillon, 2018-11-01:
I think I may have raised this issue before, but I again found myself hand-editing the json files to get the OB number correct so as not to annoy the GTC (who again moaned at me last night for one occurrence where I forgot to change the OB number). This happens, for example, when the observer massively changes the OB submitted by the PI, and then wants to repeat the observation for a second OB without having to download the new OB from the phase II system and changing all of the parameters again. It would make life much easier if there is a box in the "Next run parameters" area of hdriver that allows this number to be read, and changed if necessary.

vikdhillon, 2018-11-01:
A further request from the GTC - please could the OB number be written with 4 digits, e.g. "0003".

StuartLittlefair, 2018-11-01:
I can make the change for the GTC I think. 

Not sure that I agree about putting an OB box in hdriver. We should chat about it offline but the whole point of an OB number is so the PI can check the data against their requested setup. 

Fair enough you might know what you are doing but putting an OB box where it can be changed is a bit vulnerable to cockups in my opinion

StuartLittlefair, 2018-11-01:
To clarify my comments above, I’m not sure we should *encourage* users not to download the correct OB from the phase II. 

For example it would be too easy to miss that the second OB had a subtly different setup, or was for a shorter length of time or similar. 

Annoying as it is, I feel that downloading and editing the OB is the correct thing to do.

vikdhillon, 2018-11-01:
Hmm, I see what you mean. Well how about an expert mode option then?

StuartLittlefair, 2018-11-01:
Expert mode is a good idea. 

We should maybe think about if there’s an alternative to us editing OBs so much. It’s a bit of a conflict between the way we operate (pseudo visitor) and the way the GTC runs. It’s almost essential at the moment because some of the OBs are so wrong, but I have this nagging worry about how I’d feel if I had 8m time and it was wasted because someone changed my phase II without telling me. 😮

StuartLittlefair, 2018-11-02:
I've got a working version of this up and running - see screenshot. 

[image]

I've decided to have it so that, in non-expert mode, the ProgrammeID and OBID are disabled. They are either loaded from JSON or set by the run type. I think perhaps I will do the same for the PI. Can you see any problems with this approach  @vikdhillon?

PR #135 (2018-11-06): add runpars to list of widgets with expert mode
Closes #134 in combination with  https://github.com/HiPERCAM/hcam_widgets/pull/13.

The linked PR above adds all the functionality, but we have to add the ```RunPars``` widget to the list of those affected by expert mode.
closed 2018-11-06 by StuartLittlefair