HiPERCAM/hcam-drivers #137: Incredibly flakey slide control

vikdhillon, 2019-01-07:
On powering up HiPERCAM from cold, I tried to control the slide directly from the command line on the rack PC. Repeated slide home/reset/position commands failed with the same http connection error (apologies for not recording the exact error here, but I was up at the electronics cabinet at the time). So I tried running hdriver from the electronics cabinet. However, start_hicam failed at the hdriver stage (i.e. the various server windows opened, but hdriver did not) with an error about a url connection refused (again, apologies for not noting down the error).

I then tried controlling the slide again from the command line, and this time it worked! So starting hdriver, even though it failed to come up, obviously cleared the connection problem I was having before. However, although I could now control the slide from the command line, the slide position command always gave the home position (2700 pixels or whatever) even if it was clearly not at the home position. Weirdly, though, the slide position command did work ok whilst the slide was moving - it only reported the wrong position once it had stopped moving.

I then tried running start_hicam again, and this time it worked without any errors! I then tried moving the slide and testing the slide position command from the gui, and everything worked perfectly. So I now tried the slide position command from the command line and it also now worked perfectly, i.e. it no longer always reported its position as being 2700 pixels.

I then shut down hdriver etc on the rackmount monitor and went down to the control room and ran start_hicam there. It failed with this error, which I think is the same (unrecorded) error I mentioned above:

ngclin7 insuser:~ 1001 > start_hicam -g &
[1] 16049
ngclin7 insuser:~ 1002 > RUN:  ngcdcsStopServer IRCAM1

ngclin7 insuser:~ 1002 > RUN:  ngcdcsStartServer IRCAM1 -port 8001 -gui
RUN:  ngcbCmd off
OK

RUN:  ngcbCmd setup DET.SYSCFG hipercam_5ccd.cfg
OK

RUN:  ngcbCmd standby
OK

RUN:  ngcbCmd setup DET.FRAM.NAMING auto
OK

RUN:  ngcbCmd setup DET.FRAM.FILENAME run
OK

ls: cannot access /data/run*.fits: No such file or directory
last written frame was run0000.fits
starting HServer
starting HWServer
starting FileServer
starting GTC_Offsetter
starting GUI
Traceback (most recent call last):
  File "/usr/bin/hdriver", line 327, in <module>
    gui = GUI()
  File "/usr/bin/hdriver", line 53, in __init__
    self.globals.cpars['eso_server_online'] = isOnline(self.globals)
  File "/usr/lib/python2.7/site-packages/hcam_widgets/misc.py", line 520, in isOnline
    response = urllib.request.urlopen(url, timeout=2)
  File "/usr/lib64/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib64/python2.7/urllib2.py", line 431, in open
    response = self._open(req, data)
  File "/usr/lib64/python2.7/urllib2.py", line 449, in _open
    '_open', req)
  File "/usr/lib64/python2.7/urllib2.py", line 409, in _call_chain
    result = func(*args)
  File "/usr/lib64/python2.7/urllib2.py", line 1244, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib64/python2.7/urllib2.py", line 1214, in do_open
    raise URLError(err)
urllib2.URLError: <urlopen error [Errno 111] Connection refused>

[1]+  Done                    start_hicam -g
ngclin7 insuser:~ 1002 > 

I shut down all of the server,etc, windows and typed start_hicam again. I got the same error as above again. Then, without shutting down and of the server windows, I tried typing just hdriver and it worked! Then I shut down everything and typed start_hicam again and this time it worked.

So, in summary, it all seems rather flakey. I might be wrong, but I suspect this is all to do with the connections to the slide...

StuartLittlefair, 2019-01-07:
Hi Vik

Can’t do much about this from Greece but this is two separate problems. One is user error, the other a genuine bug. 

User error: all hardware communication now goes through a dedicated hardware server. This is necessary to speed up hardware control, and avoid GUI lockups. But it does mean the hardware server needs to be started before the slide command line will work. 

The second part - about hdriver not starting up - is a genuine bug. I’ll look at that when I get time. In the meantime I think a ```start_hicam``` followed by opening hdriver manually if it doesn’t start up should be an ok workaround.

vikdhillon, 2019-01-07:
Hi Stu,

Ah - I see - thanks for explaining about the hw_server - I hadn't
realised that. (I don't remember seeing the HWServer window pop up
before when running start_hicam - is that something you added
recently?)

Don't worry about looking into the bug - the workaround will be fine
during this run. Enjoy your hols!

V.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

closed 2019-07-06 by StuartLittlefair