HiPERCAM/hcam-drivers #138: Problem with slide

vikdhillon, 2019-01-09:
Still experiencing problems with the slide, controlling it exclusively from the gui and with the HWServer running. It worked perfectly last night, and moved fine this afternoon for taking biases. But when I went to unblock the slide in preparation for flats, I got an error when I requested the position and then when I requested it again, it told me it was at 2759.9pix (the home position), which is incorred (it was at 1100pix, the unblock position). What's weird is that the position is recorded correctly whilst moving, but once the slide gets to its position, it reports a position error and then when you try requesting the position it reports that it is at the home position! e.g.

17:06:51 - Current position =  710.9 pixels (67.9 mm, 870922 ms)
17:06:53 - Executing command: position
17:06:53 - Current position =  894.8 pixels (61.8 mm, 792736 ms)
17:06:57 - Executing command: position
17:06:57 - Error in Slide thread: ValueError: could not get position from slide: command failed - position of slide is undefined until slide homed
17:06:57 - You may want to try again; the slide is unreliable
in its error reporting. Try "position" for example
17:07:00 - Executing command: position
17:07:00 - Current position = 2759.9 pixels (0.0 mm, 136 ms)
17:09:16 - Executing command: position
17:09:16 - Current position = 2759.9 pixels (0.0 mm, 136 ms)

Whilst all of this is happening, I get the following errors in the HWServer window:

ERROR:tornado.access:500 GET /slide/position (127.0.0.1) 2.64ms

I tried shutting down the HWServer, opening a new one and got the same problem. Also tried restarting hdriver and had the same problem.

StuartLittlefair, 2019-01-09:
You didn’t click “enable” or “disable” by any chance did you? I had this when I enabled the potentiometer which was not in the default position and the command to disable it doesn’t work. In that case the slide will move to one end or the other. 

The other possibility is that this is happening because the slide position request is raising an error. This sometimes happens when the slide is busy and you request a position. As the gui sometimes says the slide error reporting is flakey (hardware) and you probably shouldn’t request the position a lot since I think it can get the slide in a state. 

To clarify what’s going on here: is the slide position actually ok? Does it eventually give a sensible answer if you leave it alone a bit?

vikdhillon, 2019-01-09:
Pretty sure I didn't click enable or disable (I wasn't in expert mode and
didn't see such options).

I only repeatedly started hitting position whilst moving the slide after
the first couple of times that it failed, so this wasn't the original cause
of the problem.

This is the same behaviour we saw a couple of days ago (once I'd got the
hwserver going) when controlling from the command line up at the
instrument. On that occasion the slide definitely was moving to position
and only reporting the correct position whilst moving. As soon as the final
position was (correctly) achieved, it always reported (incorrectly) that it
was at the home position.

Two weird things. First, this started happening today, after a whole day of
no problems yesterday. Pretty sure nothing was touched or was changed
in-between. Also, when this problem  appeared when we started up from cold
a few days ago, it somehow fixed itself. Now, despite restarting
everything, the problem won't go away. The only thing I've not tried is
power cycling the slide, but that's tricky now as the telescope is
inaccessible.

vikdhillon, 2019-01-09:
Just tested the slide.

It is definitely moving as commanded. The only problem is the position
command when the slide has reached its target position, which is returned
incorrectly as the home position.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

vikdhillon, 2019-01-09:
Ah - discovered something else.

The goto button seems to work, in that the slide moves as expected, even
though the final position is reported incorrectly (as described previously).

However, the unblock and block buttons reports an error on clicking it
(Error in Slide Thread: ValueError: could not get position from slide:
command failed - position of slide is undefined until slide homed). It is
tough to tell if the slide is actually moving in this case, although I
suspect not (the repeated position commands come up with errors in this
case).

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2019-01-09:
OK. I’m pretty sure there’s a problem here with the slide getting into a confused state but I’m not sure why. It was bombproof when James and I were out last time. 

Restarting the hardware server might help; it opens a new serial connection to the slide, but it sounds like it needs a power cycle. 

You could try the restore and reset options in expert mode if you really need to regain control. These are supposed to be the equivalents of a power cycle

closed 2019-07-06 by StuartLittlefair