HiPERCAM/hcam-drivers #139: Dithering doesn't seem to work

vikdhillon, 2019-01-09:
Doing flats tonight and dithering didn't seem to work. The first frame would be taken and then a dither command would be executed and then hdriver would not start the second exposure for some reason. No errors reported anywhere. Note that manually dithering using the gtc_tweak_pointing script works fine, so the link to the telescope is fine. Below is a copy of the last hundred or so lines of the GTC_Offsetter window, when I was repeatedly stopping and starting the exposure (after leaving it 30-60+ secs after the first frame had been taken without the second frame starting). Perhaps I should have left it waiting for longer, but I didn't want to miss the flats tonight.

starting
forcing an offset
sending tel offsets 0.0 0.0
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope ready - starting exposure
OK

sending tel offsets 0.0 10.0
sending tel offsets 10.0 0.0
sending tel offsets 0.0 -10.0
stopping
starting
forcing an offset
sending tel offsets 0.0 0.0
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope not ready yet - check again in 1s
Checking if telescope is ready
telescope ready - starting exposure
OK

stopping

vikdhillon, 2019-01-09:
This is potentially critical as we're planning on doing Trujillo's proposals tonight, which all require dithering.

vikdhillon, 2019-01-09:
This could be a telescope issue - something to do with the M1 not settling - Hector is on to it...

vikdhillon, 2019-01-09:
Hmm - weird - the gtc_offsetter code has a timeout waiting for the telescope to get to position, so why doesn't this kick in?

vikdhillon, 2019-01-09:
Not sure it is the telescope now. The timeout in gtc_offset seems to be 40secs, so for some reason this isn't being triggered.

StuartLittlefair, 2019-01-09:
This is an issue with tornado versions. Reverting to tornado 4.5.3 fixes the issue, but this is incompatible with code in the ```hwserver``` branch, which gives greatly improved hardware functionality. I will have to ask for advice from the tornado developers about the change...

StuartLittlefair, 2019-01-31:
I’ve made a suggested change in the comments of ```gtc_offsetter```. Another option is to change the ```Observer``` option to a ```PollingObserver```

StuartLittlefair, 2019-01-31:
A final possibility is that the absolute path needs using for everything in ```self.existing_runs```, rather than relying on a relative path. This should be harmless, so I've pushed this fix

StuartLittlefair, 2019-07-09:
This is redundant now that the hwserver changes no longer use tornado

closed 2019-07-09 by StuartLittlefair