HiPERCAM/hcam-drivers #143: Dithering must wait for autoguider to reacquire guide star

vikdhillon, 2019-06-05:
We've finally got a working GTC autoguider on Folded Cass E. Hector has defined the zone of avoidance for vignetting the HiPERCAM field. However, one problem we noticed tonight on Trujillo's program is that large dithers result in the odd frame having vignetting from what must be the autoguider arm. I think what might be happening is that the autoguider has to move and reacquire the guide star after large dithers (Trujillo's are 40"), and we're starting the exposure before this has completed, catching a little vignetting. So what we need is for Hector to provide a parameter that says when the autoguider has settled, a little like we have for M1, etc, at the moment. Do you think it would be possible to liaise with Hector to arrange this?

StuartLittlefair, 2019-06-05:
I'm on the case

StuartLittlefair, 2019-07-09:
This was fixed in the GTC software by Hector including the correct FoV for HiPERCAM

closed 2019-07-09 by StuartLittlefair