HiPERCAM/hcam-drivers #148: Rack temperature monitor flakey in hdriver?

vikdhillon, 2019-08-27:
The rack temperature was working fine for the past few days or two, but tonight it reports an error (nan). Weirdly, it complains about the TCS server, even though this is working fine as we're getting the RA,dec,etc values without problems. Here is the error in the command log:

00:15:18 - Run stopped
00:15:18 - Setting chips to idle
00:15:20 - Stopping offsets (if running
00:16:05 - Error in thread TCS: IOError: cannot communicate with GTC telescope server
00:16:27 - Error in thread TCS: IOError: cannot communicate with GTC telescope server
00:18:02 - Could not update temperature for RACK: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 Remote error on Server
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/hcam_widgets/misc.py", line 685, in get_hardware_value
    val = proxy.get_value(device, prop)
  File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
RuntimeError: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 +--- This exception occured remotely (Pyro) - Remote traceback:
 | Traceback (most recent call last):
 |   File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 1421, in handleRequest
 |     data = method(*vargs, **kwargs)  # this is the actual method call to the Pyro object
 |   File "/usr/bin/hwserver", line 173, in get_value
 |     raise RuntimeError('command execution failed: ' + str(err))
 | RuntimeError: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 +--- End of remote traceback

00:18:10 - Alarm on RACK acknowledged, will re-raise in 30 mins if not fixed
00:18:22 - Could not update temperature for RACK: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 Remote error on Server
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/hcam_widgets/misc.py", line 685, in get_hardware_value
    val = proxy.get_value(device, prop)
  File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
RuntimeError: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 +--- This exception occured remotely (Pyro) - Remote traceback:
 | Traceback (most recent call last):
 |   File "/usr/lib64/python2.7/site-packages/Pyro4/core.py", line 1421, in handleRequest
 |     data = method(*vargs, **kwargs)  # this is the actual method call to the Pyro object
 |   File "/usr/bin/hwserver", line 173, in get_value
 |     raise RuntimeError('command execution failed: ' + str(err))
 | RuntimeError: command execution failed: CORBA.BAD_OPERATION(0x0, CORBA.COMPLETED_NO)
 +--- End of remote traceback

00:19:09 - execCommand, command = "start"
00:19:10 - Response from server was OK

vikdhillon, 2019-08-28:
> Thanks Vik,
>
> Did you do anything to resolve this, or did it just clear itself? My
> suspicion is this is to do with two threads in hdriver trying to talk to
> the TCS server simultaneously...
>
> Stu

No, it remained like that all night. I'm not sure if it cleared up since I
left site yesterday afternoon.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2019-08-28:
Thanks Vik,

Did you do anything to resolve this, or did it just clear itself? My suspicion is this is to do with two threads in hdriver trying to talk to the TCS server simultaneously...

Stu

StuartLittlefair, 2021-06-02:
fixed in https://github.com/HiPERCAM/hcam-drivers/pull/153

closed 2021-06-02 by StuartLittlefair