HiPERCAM/hcam-drivers #150: Special set of parameters results in no data frames being written

vikdhillon, 2021-05-08:
When running OB number 5 of proposal GTCMULTIPLE3A-21A, no data frames were written. This OB requested 30 full-frame, unbinned images, each of 30.0 second exposures, with NSKIP of 10 in ugr and 1 in iz. The application can be started without any error, but no data frames are written and the frame counter on hdriver remains at zero whilst the exposure counter continues to increase. rtplot results in no data being plotted. Weirdly, an exposure time of 0, 10, or even 30.1 secsonds works perfectly well, but not 30.0 seconds.

StuartLittlefair, 2021-05-08:
This isn't going to be a problem with hdriver itself, but will be something to do with the ESO s/w. I've raised an issue there instead. We had similar problems in the past - certain setups that would cause frames to be written only occasionally. I'll look at what happened there.

Stu

trmrsh, 2021-05-09:
Vik, have you expermented with what parameters matter in terms of causing this? While we await a fix from ESO, it would be nice to know whether there is a simple way to avoid hitting this problem.