HiPERCAM/hcam-drivers #155: flat_spiral json file doesn't work

vikdhillon, 2021-06-09:
The flat_spiral.json setup does not seem to work anymore. No errors are reported - the run apparently starts (the run number increments by one and the exposure time counter starts on hdriver), but no frames are written. However, if you load the json file, disable dithering, enable it again, and load the spiral.txt file of offsets, the script works fine. So the problem is probably something to do with reading in the dither positions from the json file?

StuartLittlefair, 2021-06-10:
Hi @vikdhillon 

I noticed the same issue when we observed with David Garcia on 2021-06-07. Unfortunately I can't re-create the bug this morning. Loading the json file and starting a flat-field run works fine. 

I suspect this is a subtle timing issue. What should happen is that HiPERCAM starts a new exposure - say `run0007`. No data is taken because the NGC is waiting for a trigger. After 0.01 secs, the GTCServer turns on dithering and sends the trigger. 

What I *think* is happening is that the timing is slightly off, so that the run takes more than 0.01 seconds to start. The GTC sends the trigger **before** run0007 is created and the NGC starts waiting for the trigger. A second trigger is never sent, so the first frame never appears. 

I have addressed this by increasing that 0.01 second delay to 0.05 seconds. The downside of this is that it adds an **extra 0.05 seconds dead time to all dithered exposures.** 

For future debugging purposes, if this happens again, please check the following. 

The `GTCServer` window should show something like:

```
HiPERCAM is observing new run: run0007
dithering enabled
starting exposure
sending offsets to telescope:  0.0. 0.0
```

Meanwhile the `HServer` window should show something like:

```
2021-06-10T08:15:20+0000 cmd setup DET.RESTRICT F  DET.SPEED 0  DET.BINX1 1  DET.BINY1 1  DET.CLRCCD T  DET.NCLRS 5  DET.DUMMY 1  DET.FASTCLK 0  DET.EXPLED F  DET.GPS T  DET.INCPRSCX F  DET.INCOVSCY F  DET.NSKIPS1 0  DET.NSKIPS2 0  DET.NSKIPS3 0  DET.NSKIPS4 0  DET.NSKIPS5 0  DET.SEQ.CLKFILE hipercam.bclk  DET.SEQ1.DIT 4.0  DET.SEQ.TRIGGER T  DET.TDELAY.GUI 4000.0 : 
2021-06-10T08:15:21+0000 cmd start: 
2021-06-10T08:15:21+0000 cmd seq trigger:
```

If all those messages are present, I think the timing issue above must be the cause. If not, it's some other bug that I haven't spotted.

closed 2021-06-10 by StuartLittlefair

vikdhillon, 2021-06-10:
Hi Stu,

Thanks for looking into this. A couple of comments:

- The script definitely used to work without any problems (prior to this
June run). Interesting that we've both spotted problems with it since then.
Has anything changed recently that might affect it?
- I did look at the GTCServer window yesterday when we had the problems -
it definitely did not show the line "sending offsets to telescope: 0.0.
0.0" as I was looking out for this.

Cheers,

Vik.

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2021-06-10:
Cheers Tone. 

To answer your first question, everything has changed. The move to a centralised WAMP server is a major change that has affected almost every part of the software including the GUI and all the scripts. They should all function the same, but the internals are completely new. 

For the second part, the offsets line definitely won’t appear. What matters for debugging is whether the “trigger” command appears in the hserver window and, if so, when.

vikdhillon, 2021-06-10:
Ah, I hadn't realised quite how all-encompassing the change to the WAMP
server was!
I thought it was restricted to communication with the vac gauges,
meerstetters, slide and COMPO. But it is more than that - the telescope and
the NGC as well?

---
Prof. Vik Dhillon, Dept of Physics & Astronomy, Univ of Sheffield,
Sheffield S3 7RH, UK
+44 114 222 4528; www.vikdhillon.staff.shef.ac.uk
---

StuartLittlefair, 2021-06-10:
It is all-encompassing. Basically any communication between bits of software or hardware is now one through the WAMP server, so that includes all commands to the NGC, GTC or HiPERCAM hardware.