HiPERCAM/hcam-drivers #162: Setting CCD temperature using hdriver doesn't work if flow sensor has failed

vikdhillon, 2023-09-06:
The CCD1 flow sensor appears to have failed. As a result, the CCD warmed up to +8degC, unfortunately. To prevent sudden cooling if the flow sensor comes back to life, I decided to set the CCD1 temperature from -90degC to +8degC. I entered expert mode in hdriver to do this, entered 8 in the CCD1 temperature box in the "CCD TECs" pane, and hit return. All seemed find - I got sensible responses in the various windows. However, the set point did not change - I verified this using the new telemetry tool, as well as grafana. I then hit the "reset" button for CCD1 - this did work! It set the CCD set point to 0degC. So, for some reason, the value entered in the temperature box doesn't work. I wonder if this is because the CCD is in an error state due to the flow sensor not working? But then why was it possible to change the set point to 0degC? This is an important one to solve, as we need to do exactly this when a flow sensor fails and there is no one available at the GTC to help.

vikdhillon, 2023-09-06:
Here is some of the output from the hwserver window in response to the above setting changes:

17:27:08 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:27:09 CMM INFO    Resetting TEC on CCD1
17:27:42 CMM INFO    Updating CCD1
17:27:42 CMM INFO    desired setpoint 7
17:27:42 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:28:28 CMM INFO    Updating CCD1
17:28:28 CMM INFO    desired setpoint 7
17:28:28 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:29:27 CMM INFO    Updating CCD1
17:29:27 CMM INFO    desired setpoint 8
17:29:27 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:30:24 CMM INFO    Resetting TEC on CCD1
17:30:43 CMM INFO    Updating CCD1
17:30:43 CMM INFO    desired setpoint 7
17:30:43 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:31:52 CMM INFO    Resetting TEC on CCD1
17:32:59 CMM INFO    Updating CCD1
17:32:59 CMM INFO    desired setpoint 7
17:32:59 CMM INFO    publishing to topic hipercam.ccd1.setpoint
17:38:04 CMM INFO    Updating CCD1
17:38:04 CMM INFO    desired setpoint 8
17:38:04 CMM INFO    publishing to topic hipercam.ccd1.setpoint

vikdhillon, 2023-09-06:
And here is some of the hdriver command log output:

17:32:59 - publishing to topic hipercam.ccd1.setpoint
17:38:04 - Updating CCD1
17:38:04 - desired setpoint 8
17:38:04 - publishing to topic hipercam.ccd1.setpoint

StuartLittlefair, 2023-09-15:
This is fixed in https://github.com/HiPERCAM/hcam_devices/commit/2f7f7d9d42d6f9dc4f4c799d41559fc29ab3ec43

closed 2023-09-15 by StuartLittlefair