HiPERCAM/hcam-drivers #163: Add a time to rotator limit warning in hdriver

vikdhillon, 2023-10-20:
Apologies in advance if this is already available in hdriver, but I think it would be useful to have a time to rotator limit listed on hte TCS panel of hdriver, perhaps with an alarm as the limit is approached (or a recommendation to try unwrapping the rotator before the exposure is started). This happened to one of the observers during the last run, where he/she hit the rotator (software) limit and had to do a 360deg unwrap to get going again. I just asked the GTC if they have such a value they could make available for us on the TCS server, but they don't. So our only option is to calculate one ourselves. I believe the relevant TCS parameter is called ROTATOR (for which the limits are +/-240 degrees) and we'd need to calculate the time to limit using the INSTRPA and PARANG parameters and the relation ROTATOR+PARANG-INSTRPA=-180 (always).

StuartLittlefair, 2023-10-20:
We can calculate this ourselves I think without too much issue. 

For a star with hour angle $h$, declination $\delta$ and an observatory at latitude $\phi$, the parallactic angle is:

$$\tan q = \frac{\sin h}{\cos \delta \tan \phi - \sin \delta \cos h}$$

The hour angle increases at a rate $\dot{h} = 2\pi /T_s$ where $T_s$ is the sidereal day. This leads to a rate of change of parallactic angle given by:

$$\dot{q} = \frac{\cos \phi \cos {\rm Az}}{\cos {\rm Alt}} \dot{h}$$

which is obviously not constant, so we would need to use INSTRPA and the equation above for $\dot{q}$ to find the appropriate rotator limit from (+240, -240). Then use a root finding argument to find the limiting value of $h$ and thus time to limit.

vikdhillon, 2023-10-20:
Lovely!

But don't we still need ROTATOR+PARANG-INSTRPA=-180, as it's the value of ROTATOR that matters (this is the one that goes from -240 to +240).

Also, do we currently get values for all three of these rotator parameters from Hector's TCS server, or do we need to ask him to add some?

StuartLittlefair, 2023-10-20:
Yes, we do. We need to use $\dot{q}$ to see if PARANG is increasing or decreasing. This tells us if we are going to hit ROTATOR=240 or ROTATOR=-240 as the limit. 

Suppose it's ROTATOR=240. Then we set PARANG_LIMIT = -180 + INSTRPA - 240
and work out the value of $h$ that gives PARANG_LIMIT...

We get all these values from Hector, so it's just a case of doing the math

StuartLittlefair, 2023-10-24:
Leaving this here for now as I could do with talking to someone about how to handle root finding with angles, which wrap at some range (e.g $0 < q < 360$)

closed 2024-01-22 by StuartLittlefair