HiPERCAM/hcam-drivers #166: COMPO lens seems to require homing prior to every move

vikdhillon, 2024-06-01:
Each time we moved COMBO, the arms moved fine, but the lens never seemed to get to position, permanently reporting a "MOVING" status. We had to home COMPO and then click MOVE to make it work. For example, at the start of the night, we took bias frames, and so had COMPO in the park position. Then we took flats with the injection arm over the right-hand corner of the CCD and the pick-off arm parked - this needed a home to move to the lens. Then we went to the target and moved the pick-off arm to an angle of -58 degrees - this needed a home to move the lens. So the error occurred after having taken a run with COMPO in a particular position, and instructing it to then move to a new position. The error reported each time in the Command log and in the xterm running start_hicam was the following:

21:07:56 - Failed to move stages in COMPO: wamp.error.runtime_error: unsupported state transition on compo_lens
move not in disable,home,stop

StuartLittlefair, 2024-06-02:
I was able to replicate this behaviour - I think it is a hardware and not a software issue.

There are no issues with the software, and a direct query of the lens stage over serial gives the same position. Therefore I think the lens stage is indeed not moving and so never reaches an INPOS state. I managed to retrieve this by clicking "STOP", and "MOVE" again. After this the lens moved successfully from then on. Something to keep an eye on.

vikdhillon, 2024-06-02:
Thanks Stu.
Good to know a Stop then a Move works without having to Home it each time -
I didn't think about trying that last night. Maybe it is worth the system
first automatically executing a Stop command whenever the Move button is
pressed, before executing the move itself?
Cheers,
Vik.

StuartLittlefair, 2024-06-03:
That's a nice idea! I'll implement that after the exams are over